robotics-in-concert/rocon_app_platform #274: remove warning for no preferred rapp selection.

jihoonl, 2015-03-04:
These warnings are annoying..

```
[INFO] [WallTime: 1425446388.405242] Rapp Manager : 'rocon_apps/meow_chirp' added to the list of runnable apps.
[INFO] [WallTime: 1425446388.405373] Rapp Manager : 'rocon_apps/moo_chirp' added to the list of runnable apps.
[INFO] [WallTime: 1425446388.405505] Rapp Manager : 'rocon_apps/listener' added to the list of runnable apps.
[INFO] [WallTime: 1425446388.405653] Rapp Manager : 'rocon_apps/lion_chirp' added to the list of runnable apps.
[INFO] [WallTime: 1425446388.405784] Rapp Manager : 'rocon_apps/angry_cat_chirp' added to the list of runnable apps.
[INFO] [WallTime: 1425446388.405906] Rapp Manager : 'rocon_apps/talker' added to the list of runnable apps.
[WARN] [WallTime: 1425446388.406038] Rapp Manager : No preferred rapp for 'rocon_apps/listener'.  'rocon_apps/listener' has been selected.
[WARN] [WallTime: 1425446388.406164] Rapp Manager : No preferred rapp for 'rocon_apps/chirp'.  'rocon_apps/angry_cat_chirp' has been selected.
[WARN] [WallTime: 1425446388.406287] Rapp Manager : No preferred rapp for 'rocon_apps/talker'.  'rocon_apps/talker' has been selected.
```

stonier, 2015-03-04:
Warnings should still be there. But only if there 8s more than 1 implementation.

bit-pirate, 2015-03-06:
> Warnings should still be there. But only if there 8s more than 1 implementation.

+1

jihoonl, 2015-03-10:
Ok warn only there is more than 1 impl.

commit 2d9029d8 (2015-03-04): change warn to info closes #274
commit a7a1b965 (2015-03-17): better warning for preferred rapp selection closes #274
closed 2015-03-17 by jihoonl