robotics-in-concert/rocon_app_platform #296: error while rapp stopping in slow network

dwlee, 2015-06-10:
```
[INFO] [WallTime: 1433920381.881304] Robot State Manager : Bye Bye
[WARN] [WallTime: 1433920382.658735] Gatew Hub pinger : Error while reading from socket: ('timed out',)
[INFO] [WallTime: 1433920382.660082] Gateway : lost connection with hub, attempting to disconnect...
[INFO] [WallTime: 1433920382.661338] Gateway : disengaged connection with the hub [xavier][192.168.10.32:6380]
[INFO] [WallTime: 1433920383.426295] Rapp Manager : stopped rapp [waiterbot_rapps/pickup_delivery]'.
[ERROR] [WallTime: 1433920383.430431] Gateway : not connected to hub, aborting.
[WARN] [WallTime: 1433920383.431389] Rapp Manager : failed to cancel flips (probably remote hub intentionally went down as well) [1, not connected to hub, aborting]
[ERROR] [WallTime: 1433920383.435114] Gateway : not connected to hub, aborting.
[WARN] [WallTime: 1433920383.436041] Rapp Manager : failed to cancel flips (probably remote hub intentionally went down as well) [1, not connected to hub, aborting]
[ERROR] [WallTime: 1433920383.439708] Gateway : not connected to hub, aborting.
[WARN] [WallTime: 1433920383.440806] Rapp Manager : failed to cancel flips (probably remote hub intentionally went down as well) [1, not connected to hub, aborting]
```

jihoonl, 2015-06-10:
this happens when hub get disappeared while stopping rapp. It was test >500ms latency. It would happen very rarely.

stonier, 2015-08-14:
New design for gopher should overcome this, so moving to indigo legacy in case we have to rethink/revisit it.

stonier, 2015-08-14:
Not going to fix for indigo. Reopen this if it is it happens to become a major concern again.

closed 2015-08-14 by stonier