robotics-in-concert/rocon_app_platform #297: error while start rapp in slow network

dwlee, 2015-06-10:
```
[INFO] [WallTime: 1433920385.297735] Rapp Manager : request received to start rapp [concert_common_rapps/pickup_delivery]
concert_common_rapps/pickup_delivery
[INFO] [WallTime: 1433920385.298828] Rapp Manager : starting app 'concert_common_rapps/pickup_delivery' underneath young
[WARN] [WallTime: 1433920391.594300] Rapp Manager : ignoring uncessary remapping rule (ir, ir)
[WARN] [WallTime: 1433920391.594755] Rapp Manager : ignoring uncessary remapping rule (rgb, rgb)
[WARN] [WallTime: 1433920391.595076] Rapp Manager : ignoring uncessary remapping rule (depth, depth)
[WARN] [WallTime: 1433920391.595372] Rapp Manager : ignoring uncessary remapping rule (depth_registered, depth_registered)
[WARN] [WallTime: 1433920391.595746] Rapp Manager : ignoring uncessary remapping rule (scan, scan)
[WARN] [WallTime: 1433920391.596151] Rapp Manager : ignoring uncessary remapping rule (ir_scan, ir_scan)
[WARN] [WallTime: 1433920391.596484] Rapp Manager : ignoring uncessary remapping rule (localize, localize)
[WARN] [WallTime: 1433920391.596780] Rapp Manager : ignoring uncessary remapping rule (docking_interactor, docking_interactor)
[WARN] [WallTime: 1433920391.597080] Rapp Manager : ignoring uncessary remapping rule (delivery_order, delivery_order)
[WARN] [WallTime: 1433920391.597435] Rapp Manager : ignoring uncessary remapping rule (update_ar_pairs, update_ar_pairs)
[WARN] [WallTime: 1433920391.597743] Rapp Manager : ignoring uncessary remapping rule (localize, localize)
[WARN] [WallTime: 1433920391.598036] Rapp Manager : ignoring uncessary remapping rule (initialpose, initialpose)
[WARN] [WallTime: 1433920391.598355] Rapp Manager : ignoring uncessary remapping rule (scan, scan)
[WARN] [WallTime: 1433920391.598638] Rapp Manager : ignoring uncessary remapping rule (initialpose, initialpose)
[WARN] [WallTime: 1433920391.598917] Rapp Manager : Remapping rule for map already exists. Ignoring remapping rule from map to /map
[WARN] [WallTime: 1433920391.599224] Rapp Manager : ignoring uncessary remapping rule (navigator, navigator)
[WARN] [WallTime: 1433920391.599579] Rapp Manager : ignoring uncessary remapping rule (move_base, move_base)
[WARN] [WallTime: 1433920391.599869] Rapp Manager : Remapping rule for map already exists. Ignoring remapping rule from map to /map
started roslaunch server http://young:36025/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.13
 * /young/amcl/base_frame_id: base_footprint
 * /young/amcl/global_frame_id: map
 * /young/amcl/gui_publish_rate: 10.0
 * /young/amcl/initial_pose_a: 0.0
 * /young/amcl/initial_pose_x: 0.0
 * /young/amcl/initial_pose_y: 0.0
 * /young/amcl/kld_err: 0.05
 * /young/amcl/kld_z: 0.99
 * /young/amcl/laser_lambda_short: 0.1
 * /young/amcl/laser_likelihood_max_dist: 2.0
 * /young/amcl/laser_max_beams: 60
 * /young/amcl/laser_max_range: 12.0
 * /young/amcl/laser_model_type: likelihood_field
 * /young/amcl/laser_sigma_hit: 0.2
 * /young/amcl/laser_z_hit: 0.5
 * /young/amcl/laser_z_max: 0.05
 * /young/amcl/laser_z_rand: 0.5
 * /young/amcl/laser_z_short: 0.05
 * /young/amcl/max_particles: 2000
 * /young/amcl/min_particles: 500
 * /young/amcl/odom_alpha1: 0.2
 * /young/amcl/odom_alpha2: 0.2
 * /young/amcl/odom_alpha3: 0.2
 * /young/amcl/odom_alpha4: 0.2
 * /young/amcl/odom_alpha5: 0.1
 * /young/amcl/odom_frame_id: odom
 * /young/amcl/odom_model_type: diff
 * /young/amcl/recovery_alpha_fast: 0.0
 * /young/amcl/recovery_alpha_slow: 0.0
 * /young/amcl/resample_interval: 1
 * /young/amcl/transform_tolerance: 1.0
 * /young/amcl/update_min_a: 0.2
 * /young/amcl/update_min_d: 0.25
 * /young/amcl/use_map_topic: True
 * /young/ar_pair/target_offset: 0.5
 * /young/ar_pair_tracking/base_frame: base_footprint
 * /young/ar_pair_tracking/global_frame: map
 * /young/ar_pair_tracking/marker_frame: camera_rgb_optica...
 * /young/ar_pair_tracking/publish_transforms: False
 * /young/ar_track_alvar/max_frequency: 4.0
[... 328 lines cut ...]
process[young/docking_interactor-36]: started with pid [24957]
[ INFO] [1433920398.807867648]: /young/docking_interactor : Initialized
[ INFO] [1433920398.817672446]: /young/docking_interactor : Wait for docking ar tracker
process[young/dock_drive-37]: started with pid [25016]
[ INFO] [1433920399.335622633]: Loading nodelet /young/dock_drive of type kobuki_auto_docking/AutoDockingNodelet to manager /mobile_base_nodelet_manager with the following remappings:
[ INFO] [1433920399.335761615]: /young/delivery_order -> /delivery_order_fb3df4430ffd42399d5e745c8b941387
[ INFO] [1433920399.335814504]: /young/dock_drive/core -> /mobile_base/sensors/core
[ INFO] [1433920399.335854833]: /young/dock_drive/dock_ir -> /mobile_base/sensors/dock_ir
[ INFO] [1433920399.335896600]: /young/dock_drive/motor_power -> /mobile_base/commands/motor_power
[ INFO] [1433920399.335947287]: /young/dock_drive/odom -> /odom
[ INFO] [1433920399.335995766]: /young/dock_drive/velocity -> /cmd_vel_mux/input/auto_docking
[ INFO] [1433920399.336034806]: /young/map -> /map
[ INFO] [1433920399.336069425]: /young/map_metadata -> /map_metadata
[ INFO] [1433920399.336128269]: /young/marker_pose_list -> /annotation/ar_markers
[ INFO] [1433920399.336172581]: /young/table_pose_list -> /annotation/tables
[ INFO] [1433920399.336211634]: /young/viz_marker_pose_list -> /annotation/viz_tables
process[young/robot_pose_publisher-38]: started with pid [25088]
[INFO] [WallTime: 1433920399.938567] Rapp Manager : xavier281cec8df1ce47e8b156403feeea14bd
[ERROR] [WallTime: 1433920400.443481] Gateway : not connected to hub, aborting.
[ERROR] [WallTime: 1433920400.444368] Rapp Manager : failed to flip [1, not connected to hub, aborting]
[ERROR] [WallTime: 1433920400.447759] Gateway : not connected to hub, aborting.
[ERROR] [WallTime: 1433920400.448496] Rapp Manager : failed to flip [1, not connected to hub, aborting]
[ERROR] [WallTime: 1433920400.451451] Gateway : not connected to hub, aborting.
[ERROR] [WallTime: 1433920400.452148] Rapp Manager : failed to flip [1, not connected to hub, aborting]
```

jihoonl, 2015-06-10:
this happens when hub get disappeared while starting rapp.  It requires update in rapp manager and scheduler both to fix the problem.

It was test >500ms latency. It would happen very rarely.

jihoonl, 2015-06-15:
[here](https://github.com/robotics-in-concert/rocon_app_platform/blob/indigo/rocon_app_manager/src/rocon_app_manager/rapp_manager.py#L601-L606) does not handle the flip failure case. 

It should stop rapp if it fails after a couple of flip attempts

commit 28edbc6b (2015-06-15): attempts to flip public interface a couple of times. Then stopps rapp if it fails after all attempts. closes #297
closed 2015-06-15 by jihoonl