robotics-in-concert/rocon_app_platform #300: rocon_app list command shows Mulitple rapps with same name error

dhirajdhule, 2015-06-20:
I had installed rocon from debs intitially, then I decided to install it from sources, so I removed all debs using

sudo apt-get purge ros-indigo-rocon-* 
i also removed kobuki_rapps package using
sudo apt-get purge ros-indigo-kobuki-* 
 
I installed rocon from sources as described in tutorial on rocon wiki page

when I am doing rocon_app list, I am getting an error of 

Multiple packages found with the same name "kobuki_rapps":
- home/flight/rocon/src/rocon_rapps/kobuki_rapps
- opt/ros/indigo/share/kobuki_rapps

Shall I do anything like updating the rocon_app_platforms cache or something?
Or am I missing some crucial step after removing debs?

btw similar error didnt let me install kobuki using tutorial on http://wiki.ros.org/turtlebot/Tutorials/indigo/Turtlebot%20Installation
but that was obvious, as it clashed with existing kobuki_rapps package in my rocon workspace.
But this error  
And all of above directories are permanently sourced.

dhirajdhule, 2015-06-21:
Got the problem...
I had indexed rapps when I had multiple rapps...which I am not sure how is possible( two packages with same name)

when I deleted files in ~/.ros/rocon/rapp/ it worked.

closed 2015-06-21 by dhirajdhule

jihoonl, 2015-06-24:
Hm.. rapp directory is used only when you utilises rapp repository.

Did you try to use [rapp indexer](http://wiki.ros.org/rocon_app_utilities/Tutorials/indigo/Create%20a%20Rapp%20Index)?