robotics-in-concert/rocon_app_platform #308: [gopher] Flip rules for subscribers

asmodehn, 2015-10-15:
Added this for the automatic door integration to work by flipping subscribers from inside a rapp : 
https://github.com/robotics-in-concert/rocon_app_platform/commit/f079f7858d7b206b3e22c164712e32f1948f3f16

Not sure if it is the best place to do it, and how the public.interface relate to that at the moment in the gopher branch...
@stonier : In need of some review...

stonier, 2015-10-19:
Ok...looks like I threw that in there to get started with publishers because that's all we were using.

Now is the time to add subscribers, services, action clients and action servers.

stonier, 2015-10-19:
Just adding subscribers, and just for the concert namespace (not the app namespace) isn't sufficient.

commit 7f97ce9c (2015-10-19): [rocon_app_manager] concert client should share all connection types, not just publishers, fixes #308.
closed 2015-10-22 by stonier