robotics-in-concert/rocon_app_platform #313: rocon app manager crashing

tfoote, 2016-06-27:
Running turtlebot_bringup minimal.launch on jessie.
```
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/rocon_app_manager/standalone.py", line 19, in <module>
    standalone_rapp_manager = rocon_app_manager.Standalone()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rocon_app_manager/standalone.py", line 105, in __init__
    self.parameters = StandaloneParameters()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rocon_app_manager/ros_parameters.py", line 79, in __init__
    for pkg, preferred_rapps_dict in preferred.iteritems():
AttributeError: 'list' object has no attribute 'iteritems'
[rocon/app_manager-9] process has died [pid 7188, exit code 1, cmd /opt/ros/kinetic/lib/rocon_app_manager/standalone.py __name:=app_manager __log:=/home/linaro/.ros/log/7515dd58-3ca2-11e6-b4cf-0200b76b1d40/rocon-app_manager-9.log].
log file: /home/linaro/.ros/log/7515dd58-3ca2-11e6-b4cf-0200b76b1d40/rocon-app_manager-9*.log
```

This line appears to be the problem: https://github.com/robotics-in-concert/rocon_app_platform/blob/gopher-devel/rocon_app_manager/src/rocon_app_manager/ros_parameters.py#L76

SouLeo, 2016-11-01:
I have the same error? Did you find a solution?

jcpince, 2016-11-08:
The problem is that "preferred = rospy.get_param('~preferred', [])" returns a list instead of a dictionary.
Given that it returns a list containing 1 dictionary, the parameter server is started but the interface between it and our rocon app is not as expected; it should return a dictionary storing dictionaries as values...

vitorhirozawa, 2017-08-24:
How can I solve this? I am new in python.

jcpince, 2017-08-28:
Given that I get a dictionary when running "roslaunch rocon_app_manager standalone.launch", I assume that you might not use preferred_default.yaml and then fall back on the default value when running "rospy.get_param('~preferred', [])" which returns a list... Changing the line into rospy.get_param('~preferred', {}) should be ok.

vitorhirozawa, 2017-08-28:
Hi @jcpince, thank you for your response.

I already tried this modification but I got the same error message.

I use ROS Kinetic and "preferred_default.yaml" is not used.
I try "roslaunch turtlebot_bringup concert_minimal.launch" and I have:
...
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/rocon_app_manager/standalone.py", line 19, in <module>
    standalone_rapp_manager = rocon_app_manager.Standalone()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rocon_app_manager/standalone.py", line 105, in __init__
    self.parameters = StandaloneParameters()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rocon_app_manager/ros_parameters.py", line 76, in __init__
    for pkg, preferred_rapps_dict in preferred.iteritems():
AttributeError: 'list' object has no attribute 'iteritems'
[ INFO] [1503946266.627515496]: Zeroconf: service successfully established [turtlebot][_ros-master._tcp][11311]
[rocon/app_manager-9] process has died [pid 14204, exit code 1, cmd /opt/ros/kinetic/lib/rocon_app_manager/standalone.py __name:=app_manager __
...

Is there a solution?

jcpince, 2017-08-28:
I'm using ros kinetic as well (I'm testing with the ubuntu packages).

I added a log this morning right before the for loop:
        self.preferred = {}
        print("JCDBG: ", preferred)
        for pkg, preferred_rapps_dict in preferred.iteritems():
...

And I got the following log:
./ad2b5eb0-8c02-11e7-8439-b88a60b1a34b/rocon-app_manager-2-stdout.log:('JCDBG: ', {'rocon_apps': {'chirp': 'rocon_apps/moo_chirp', 'talker': 'rocon_apps/talker'}})

So, this is a dictionary and it's coming from my /opt/ros/kinetic/share/rocon_app_manager/param/preferred_default.yaml.

Can you apply the same log and paste the output?

vitorhirozawa, 2017-08-29:
Hi @jcpince, thanks for your help.

I added to "ros_parameters.py":
...
self.preferred = {}
print("OI GATINHAs: ", preferred)
for pkg, preferred_rapps_dict in preferred.iteritems():
...

I got the same error message and the logfile "rocon-app_manager-9-stdout.log" has this content:
('OI GATINHAs: ', [{'rocon_apps/chirp': 'rocon_apps/moo_chirp'}])

My "/opt/ros/kinetic/share/rocon_app_manager/param/preferred_default.yaml" is:
#Uncomment and modify below to explicitly set the preferred
#implementation for an ancestor rapp specification:
preferred:
  #when there are several child implementations 
  'rocon_apps/chirp': 'rocon_apps/moo_chirp'
  #when an ancestor is an implementation and you specifically want the ancestor
  'rocon_apps/talker': 'rocon_apps/talker'

jcpince, 2017-08-29:
You should have spaces starting all the lines after preferred...

I think that your yaml file syntax is not correct...

Something like:
# Uncomment and modify below to explicitly set the preferred
# implementation for an ancestor rapp specification:
preferred:
  # when there are several child implementations
  'rocon_apps/chirp': 'rocon_apps/moo_chirp'
  # when an ancestor is an implementation and you specifically want the ancestor
  'rocon_apps/talker': 'rocon_apps/talker'

vitorhirozawa, 2017-08-29:
Sorry I edit the yaml code here on github because I clicked on "Preview" and the formatting was strange.

Here is the code without edit:
`# Uncomment and modify below to explicitly set the preferred
# implementation for an ancestor rapp specification:
preferred:
  # when there are several child implementations
  'rocon_apps/chirp': 'rocon_apps/moo_chirp'
  # when an ancestor is an implementation and you specifically want the ancestor
  'rocon_apps/talker': 'rocon_apps/talker'`

jcpince, 2017-08-29:
At some point I was able to reproduce your issue but after reinstalling the turtlebot_bringup packages and quickly setting manually the environment variables.

So, there is a bug but I think that you should go carefully through the tutorial to get it working fine. Fixing this bug will likely lead you to another issue if your environment is not set properly...

fowie, 2017-09-15:
I'm seeing the same issue.  Ubuntu 16.04.  Installed ROS from debs and trying to install turtlebot from source a la http://wiki.ros.org/turtlebot/Tutorials/indigo/Turtlebot%20Installation section 1.3 but changing "indigo" to "kinect".  Everything builds just fine but I still see the same rror on line 76 of ros_parameters.py and my yaml code looks identical to vitorhirozawa.  I've tried catkin_clean on rocon, kobuki, and turtlebot and rebuilding all of them, same issue.  I've also tried just installing ros-kinetic-rocon-xxx packages and running it that way and I see the same error, so I don't think it's a build issue.